Open-Acidification/AlkalinityTitrator #16: Refactor to allow for mocking the ADC

prestoncarman, 2021-07-28:


KonradMcClure, 2021-08-11:
The titrator uses an ADS1115 Analog to Digital Converter to read the voltage on the glass pH probe. We need to create a mock class for the ADS1115 to simulate the connection for running the program on a device other than the titrator. Currently, the program will crash when it reaches any code using the ADS1115, primarily interfaces.py

KonradMcClure, 2021-08-11:
The ADS1115 has been mocked in ads_mock.py, which simply takes the same inputs on init. However while a `pH_Probe` class was created, which imports the actual ADS1115 module for  it's `__init__()`, it turned out it wasn't really necessary for the `ph_probe_mock` class to have a mocked version. So it technically exists, but can probably be removed later unless we decide it's needed.

closed 2021-08-11 by KonradMcClure