Open-Acidification/AlkalinityTitrator #29: Remove legacy use of IS_TEST

KonradMcClure, 2021-08-11:
`IS_TEST` was originally used for a testing mode that was on the titrator itself. It would be set to `True` when the ADC couldn't be contacted. Afterwards, some of the options in the menu had alternate test versions that would run if IS_TEST was true with fake readings printed.

This old use is now redundant now that some hardware mock classes have been created and needs to be removed, as there are still remnants of it throughout the program.

KonradMcClure, 2021-08-13:
The last legacy use of it is in routines.py, `user_choice == 4` where Test Mode is toggled on and off. This doesn't work the way it was originally intended and can be removed. Unsure if removing it entirely, changing to some other function, or what is the best option there.

TaylorSmith28, 2023-02-28:
This has been fixed with #90