Open-Acidification/AlkalinityTitrator #30: Rename temp to Temperature throughout

KonradMcClure, 2021-08-11:
I used temp as a shorthand for temperature (i.e. `temp_probe`, `tempcontroller`), but given we aren't limited on characters and temp is also common for temporary, it should be changed to spell out temperature to be self-documenting.

d-cryptic, 2021-08-12:
Hey @KonradMcClure I want to contribute to this issue

KonradMcClure, 2021-08-12:
Hi @d-cryptic, feel free to create a fork, create a branch off of main, make the edits, and then submit a Pull Request. This is a nice first issue to work on since you'll get a quick look at the various files.

NOTE: I currently have Pull Request #38 in that reorganizes files to match what's in Issue #37. It might be a good idea to wait until that's merged to start on this issue. We're having issues with some of the linters we're using throwing errors, so once that's figured out it should be merged.

KonradMcClure, 2021-08-13:
@d-cryptic, the PR has been merged now, so you should be good to contribute if you'd like to!

d-cryptic, 2021-08-13:
Yes, I want to contribute.

KonradMcClure, 2021-08-13:
Great, let me know if you have any questions

d-cryptic, 2021-08-13:
I have two questions:
1. I have to change temp to temperature in every file right?
2. Should I create PR for each file separately or create PR for all the files at once?

d-cryptic, 2021-08-13:
And also I dont have raspberry pi or arduino to test the project, I thought that only changing the variable names might not require it, will it be ok?

KonradMcClure, 2021-08-13:
> I have to change temp to temperature in every file right?
More or less. I wouldn't necessarily do a mass find-and-replace of temp to temperature, but rather CTRL+F and check each one to discern if it needs to be changed or not. One reason for that is I know for some of the `setup_<module>()` functions (`setup_lcd()`, `setup_keypad`, etc) in `interfaces.py`, `temp` is used to mean "temporary". In 90% of cases though, "temp" is referring to "temperature".

> Should I create PR for each file separately or create PR for all the files at once?
All at once is better. When the PR is submitted I'll go through the diff and review it.

> And also I dont have raspberry pi or arduino to test the project, I thought that only changing the variable names might not require it, will it be ok?
My main task I've been working on lately is getting it to run without the need for a raspberry pi. If you run the program from the `./AlkalinityTitrator/` directory as `python .\src\titration.py -test` it will start up in a test mode that will display in the console. That way you can make sure it doesn't break anything. In addition, running `python -m pytest -vv` from the root folder will run a series of tests that'll also be good for checking if anything was broken by a change.

d-cryptic, 2021-08-13:
ok, I will start working on this

d-cryptic, 2021-08-14:
Hey @KonradMcClure I have replaced all 'temp' with 'temperature' except some of temps in interfaces.py as you mentioned above. I have initiated a pull request, please review it. Also I was unable to tests the project locally as I was getting some issues/errors, I will try again to run the project locally maybe tomorrow.

closed 2021-08-16 by KonradMcClure