Open-Acidification/AlkalinityTitrator #31: Implement stir control

KonradMcClure, 2021-08-11:
We have a custom stir control assembly and now that the housing will start to be worked on, it's about time to finally implement stir control into the program.

The stir control is handled by a [BD65496MUV DC Motor Controller breakout board from Pololu](https://www.pololu.com/product/2960). The Raspberry Pi's `GPIO13` or `board.D13` pin is hooked to the PWM input of the board, and `GPIO11` or `board.D13` is connected to the Motor Direction pin. There was no specific use-case for the Motor Direction pin, but we had free GPIO pins so we figured why not. Could possibly be used for a sort of breaking feature when we stop.

To Do:
* **Make a class** for the stir control, i.e. `stir_control`
* **Implement basic methods** of `start`, `set_speed_slow`, `set_speed_fast`, `set_speed` (perhaps a custom speed with a PWM value), `stop`, etc. 
* **Create a mock version**. This would mainly be for a partially mocked titration device where we just want to test things without having the actual stir motor go crazy. 
* Create test routines

KonradMcClure, 2021-08-19:
This has been started on my `dev_titrator_testing` branch now that I have my hands on the device at Rosario

TaylorSmith28, 2023-02-28:
This has been fixed with #89

closed 2023-02-28 by TaylorSmith28