Open-Acidification/AlkalinityTitrator #50: Implement proper arduino/syringe pump class

KonradMcClure, 2021-08-17:
Currently the arduino is represented by a simple serial connection (or mock of it). One thing that would make things a bit better, especially for the mock and testing, would be to create a proper `arduino `or `syringe_pump` class that wraps this `Serial` class inside it. A few reasons for this:
1. All of the pump functions down to the lowest level are completely handled within `interfaces.py`. This is unusual considering the other classes such as `LCD`, where there are functions in `interfaces.py` that talk to the `LCD` but the `LCD` class handled the low level functionality itself.
2. One particular feature that's slowing things down is the wait time. When the Raspberry Pi tells the Arduino to drive the stepper motor in `interfaces.drive_step_stick()`, a `wait_time` is calculated based on the speed of the motor, and the Raspberry Pi waits that long before continuing to make sure the pump finishes. Currently, without explicitly checking if the arduino is the mock, that wait time is pointless if we're in testing mode. If these functions were wrapped up in a larger class, the wait time on the mock could be 0 and things would speed up a bit. This effects both how fast pytest runs and how much time I spend staring at the terminal as I inch closer to my ultimate demise.

KonradMcClure, 2021-08-17:
Finished this up in my `dev_arduino_class` branch with a new Syringe_Pump class and mock. After #49 I'll do a new PR for that

TaylorSmith28, 2023-02-27:
Fixed in #89

closed 2023-02-27 by TaylorSmith28